microsoft/AirSim-NeurIPS2019-Drone-Racing #117: How can I run the log_monitor.py?

surfii3z, 2019-11-09:
So what I did is
`~/AirSim_Training $ ./AirSimExe.sh -windowed -opengl4`
Then I run log_monitor.py in 
`~/AirSim_Training/AirSimExe/Saved/Logs$ python3.7 log_monitor.py`
Then I run baseline_racer.py in
`~/AirSim-NeurIPS2019-Drone-Racing/baselines$ python3.7 baseline_racer.py`
I got this error

> ERROR Bad line: [2019.11.09-15.02.11:244][652]LogAudioMixer: Display: No commands were queued while flushing the source manager.

Tokens: ['[2019.11.09-15.02.11:244][652]LogAudioMixer:', 'Display:', 'No', 'commands', 'were', 'queued', 'while', 'flushing', 'the', 'source', 'manager.']
ERROR Bad line: [2019.11.09-15.02.11:273][652]LogSlate: Request Window 'AirSimExe (64-bit Development GLSL_430) ' being destroyed

Tokens: ['[2019.11.09-15.02.11:273][652]LogSlate:', 'Request', 'Window', "'AirSimExe", '(64-bit', 'Development', 'GLSL_430)', "'", 'being', 'destroyed']
ERROR Bad line: [2019.11.09-15.02.11:321][652]LogSlate: Window 'AirSimExe (64-bit Development GLSL_430) ' being destroyed

Tokens: ['[2019.11.09-15.02.11:321][652]LogSlate:', 'Window', "'AirSimExe", '(64-bit', 'Development', 'GLSL_430)', "'", 'being', 'destroyed']
ERROR Bad line: [2019.11.09-15.02.11:324][652]LogSlate: Slate User Destroyed.  User Index 0, Is Virtual User: 0

Tokens: ['[2019.11.09-15.02.11:324][652]LogSlate:', 'Slate', 'User', 'Destroyed.', 'User', 'Index', '0,', 'Is', 'Virtual', 'User:', '0']
ERROR Bad line: [2019.11.09-15.02.11:324][652]LogExit: Preparing to exit.

Tokens: ['[2019.11.09-15.02.11:324][652]LogExit:', 'Preparing', 'to', 'exit.']
ERROR Bad line: [2019.11.09-15.02.11:327][652]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden

Tokens: ['[2019.11.09-15.02.11:327][652]LogSlate:', 'Updating', 'window', 'title', 'bar', 'state:', 'overlay', 'mode,', 'drag', 'disabled,', 'window', 'buttons', 'hidden,', 'title', 'bar', 'hidden']
ERROR Bad line: [2019.11.09-15.02.11:412][652]LogRendererCore: FGlobalReadBuffer::ReleaseRHI

Tokens: ['[2019.11.09-15.02.11:412][652]LogRendererCore:', 'FGlobalReadBuffer::ReleaseRHI']
ERROR Bad line: [2019.11.09-15.02.11:412][652]LogRendererCore: FGlobalDynamicReadBuffer::Cleanup()

Tokens: ['[2019.11.09-15.02.11:412][652]LogRendererCore:', 'FGlobalDynamicReadBuffer::Cleanup()']
ERROR Bad line: [2019.11.09-15.02.11:412][652]LogRendererCore: FGlobalReadBuffer::ReleaseRHI

Tokens: ['[2019.11.09-15.02.11:412][652]LogRendererCore:', 'FGlobalReadBuffer::ReleaseRHI']
ERROR Bad line: [2019.11.09-15.02.11:412][652]LogRendererCore: FGlobalDynamicReadBuffer::Cleanup()

Tokens: ['[2019.11.09-15.02.11:412][652]LogRendererCore:', 'FGlobalDynamicReadBuffer::Cleanup()']
ERROR Bad line: [2019.11.09-15.02.11:412][652]LogRendererCore: FGlobalReadBuffer::ReleaseRHI

Tokens: ['[2019.11.09-15.02.11:412][652]LogRendererCore:', 'FGlobalReadBuffer::ReleaseRHI']
ERROR Bad line: [2019.11.09-15.02.11:412][652]LogRendererCore: FGlobalDynamicReadBuffer::Cleanup()

Tokens: ['[2019.11.09-15.02.11:412][652]LogRendererCore:', 'FGlobalDynamicReadBuffer::Cleanup()']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module AutomationController (194)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'AutomationController', '(194)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module AutomationWorker (192)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'AutomationWorker', '(192)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module Voice (190)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'Voice', '(190)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module AIModule (188)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'AIModule', '(188)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module GameplayDebugger (187)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'GameplayDebugger', '(187)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module NavigationSystem (185)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'NavigationSystem', '(185)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module WebMMediaFactory (182)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'WebMMediaFactory', '(182)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module WebMMedia (180)

Tokens: ['[2019.11.09-15.02.11:457][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'WebMMedia', '(180)']
ERROR Bad line: [2019.11.09-15.02.11:457][652]LogModuleManager: Shutting down and abandoning module WmfMediaFactory (178)

[... 178 lines cut ...]
ERROR Bad line: [2019.11.09-15.02.11:505][652]LogModuleManager: Shutting down and abandoning module ImageWrapper (56)

Tokens: ['[2019.11.09-15.02.11:505][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'ImageWrapper', '(56)']
ERROR Bad line: [2019.11.09-15.02.11:505][652]LogModuleManager: Shutting down and abandoning module OpenGLDrv (54)

Tokens: ['[2019.11.09-15.02.11:505][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'OpenGLDrv', '(54)']
ERROR Bad line: [2019.11.09-15.02.11:505][652]LogModuleManager: Shutting down and abandoning module InputCore (52)

Tokens: ['[2019.11.09-15.02.11:505][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'InputCore', '(52)']
ERROR Bad line: [2019.11.09-15.02.11:505][652]LogModuleManager: Shutting down and abandoning module PhysXCooking (50)

Tokens: ['[2019.11.09-15.02.11:505][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'PhysXCooking', '(50)']
ERROR Bad line: [2019.11.09-15.02.11:505][652]LogModuleManager: Shutting down and abandoning module SteamVR (48)

Tokens: ['[2019.11.09-15.02.11:505][652]LogModuleManager:', 'Shutting', 'down', 'and', 'abandoning', 'module', 'SteamVR', '(48)']
Traceback (most recent call last):
  File "log_monitor.py", line 62, in <module>
    main() 
  File "log_monitor.py", line 59, in main
    process(line)
  File "log_monitor.py", line 31, in process
    if tokens[3] == "disqualified" and tokens[4] == '1' and tokens[0] not in disqualified_racers:
IndexError: list index out of range

I expected to use the information to keep track that what is the next gate that "drone_1" needs to pass through. Is it possible to do this?

madratman, 2019-11-10:
Run the baseline_racer.py first, then log_monitor.py. 
The call to start_race() starts the call to the server side log file generator.

surfii3z, 2019-11-11:
Thank you so much. Does the log information e.g. drone_1 pass gate 2 allow in Qualification?

madratman, 2019-11-11:
Yes, you can use it. You can also do something like https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/79#issuecomment-541883689

closed 2019-11-11 by madratman