microsoft/AirSim-NeurIPS2019-Drone-Racing #120: Is there a way to specify the size of images without modifying setting.json?

yannbouteiller, 2019-11-10:
Hello,

I will definitely not need 320*240 images, and I would like to get smaller ones directly from the simulator. Is this possible in the API without modifying the default .json file that will be used, please?

madratman, 2019-11-10:
No, settings.json is the way to go. Is modifying the settings.json file proving to be cumbersome?
If so, you can modify [this line](https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/utils.py#L93) and call geneate_settings.py

yannbouteiller, 2019-11-11:
No it is fine for me, I was just thinking about how to do this in the competition since I suppose the default settings.json will be used. But I can as well get 240*320 images and resize them.

closed 2019-11-11 by yannbouteiller