microsoft/AirSim-NeurIPS2019-Drone-Racing #121: Crash on retreiving images?

yannbouteiller, 2019-11-11:
Hello,

In my trainings for tier3 I get this crash very often:

> (pid=24631) Failed to find symbol file, expected location:
> (pid=24631) "/home/yann/Desktop/GoD/Algorithm/environments/gym-airsimneurips/gym_airsimneurips/envs/AirSim/AirSimExe/Binaries/Linux/AirSimExe.sym"
> (pid=24631) Signal 11 caught.
> (pid=24631) Malloc Size=65538 LargeMemoryPoolOffset=65554 
> (pid=24631) Malloc Size=65535 LargeMemoryPoolOffset=131119 
> (pid=24631) Malloc Size=98832 LargeMemoryPoolOffset=229968 
> (pid=24631) 9:969][967]LogPackageName: SearchForPackageOnDisk took   0.089s to resolve Soccer_Field_Easy.
> (pid=24631) [2019.11.11-06.05.29:971][967]LogStreaming: Display: ULevelStreaming::RequestLevel(/MSBuild2018/Maps/Soccer_Field_Easy_LevelInstance_2) is flushing async loading
> (pid=24631) [2019.11.11-06.05.29:971][967]LogStreaming: Display: Flushing async loaders.
> (pid=24631) [2019.11.11-06.05.30:420][969]LogUObjectHash: Compacting FUObjectHashTables data took   3.44ms
> (pid=24631) [2019.11.11-06.05.30:970][192]LogTemp: spawning player
> (pid=24631) [2019.11.11-06.05.30:970][192]LogLevel: WORLD TRANSLATION BEGIN {0, 0, 0} -> {-1810, -860, 172}
> (pid=24631) [2019.11.11-06.05.30:971][192]LogLevel: WORLD TRANSLATION END {-1810, -860, 172} took 0.6022 ms
> (pid=24631) [2019.11.11-06.05.31:971][100]LogUMG: Display: Widget Class RaceUI_C - Loaded Fast Template.
> (pid=24631) [2019.11.11-06.05.34:013][150]LogRenderer: Reallocating scene render targets to support 320x240 Format 10 NumSamples 1 (Frame:1).
> (pid=24631) [2019.11.11-06.05.34:585][593]LogUMG: Display: Widget Class RacerStatusInfo_C - Loaded Fast Template.
> (pid=24631) CommonUnixCrashHandler: Signal=11
> (pid=24631) [2019.11.11-06.05.40:982][562]LogCore: === Critical error: ===
> (pid=24631) Unhandled Exception: SIGSEGV: unaligned memory access (SIMD vectors?)
> (pid=24631) 
> (pid=24631) [2019.11.11-06.05.40:983][562]LogCore: Fatal error!
> (pid=24631) 
> (pid=24631) 0x00000000027100c7 AirSimExe!UnknownFunction(0x25100c7)
> (pid=24631) 0x00000000026a4046 AirSimExe!UnknownFunction(0x24a4045)
> (pid=24631) 0x00000000026a4a26 AirSimExe!UnknownFunction(0x24a4a25)
> (pid=24631) 0x0000000002595092 AirSimExe!UnknownFunction(0x2395091)
> (pid=24631) 0x0000000004ee8dea AirSimExe!UnknownFunction(0x4ce8de9)
> (pid=24631) 0x000000000563cf2b AirSimExe!UnknownFunction(0x543cf2a)
> (pid=24631) 0x0000000004e97661 AirSimExe!UnknownFunction(0x4c97660)
> (pid=24631) 0x0000000004e9bab8 AirSimExe!UnknownFunction(0x4c9bab7)
> (pid=24631) 0x000000000258664b AirSimExe!UnknownFunction(0x238664a)
> (pid=24631) 0x000000000258cf5a AirSimExe!UnknownFunction(0x238cf59)
> (pid=24631) 0x0000000005dac82f AirSimExe!UnknownFunction(0x5bac82e)
> (pid=24631) 0x00007fa67dc0bb97 libc.so.6!__libc_start_main(+0xe6)
> (pid=24631) 0x000000000257c029 AirSimExe!UnknownFunction(0x237c028)
> (pid=24631) 
> (pid=24631) [2019.11.11-06.05.40:996][562]LogExit: Executing StaticShutdownAfterError
> (pid=24631) [2019.11.11-06.05.41:008][562]LogCore: Warning: Unable to statfs('/home/yann/Desktop/GoD/Algorithm/environments/gym-airsimneurips/gym_airsimneurips/envs/AirSim/AirSimExe/Saved/Crashes/crashinfo-AirSimExe-pid-24873-02C6787908D76643456F2E66EC6209DC/AirSimExe.log'): errno=2 (No such file or directory)
> (pid=24631) Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> (pid=24631) Malloc Size=78456 LargeMemoryPoolOffset=308440 
> (pid=24631) Segmentation fault (core dumped)

What does it mean please? Apparently it has something to do with image sizes? I get the images in size 240*320 here.

madratman, 2019-11-11:
Can you give us a reproducible script? What's the minimal way to repro this? Just call simGetImages for a long time?

yannbouteiller, 2019-11-11:
Actually this issue appears only when we launch AirSimExe.sh with the -BENCHMARK keyword, that I was using because I don't know how to set t.maxFPS when I don't have access to the window. Is there another way of doing this?

madratman, 2019-11-11:
Well, I haven't tried that particular arg before but it doesn't sound good. Render thread is separate from game thread. I am not sure how unreal plays out if you force it to use a single thread.

closed 2019-11-11 by madratman