microsoft/AirSim-NeurIPS2019-Drone-Racing #126: Is there any function to get the race scores?

kavehkamali, 2019-11-14:
I need to extract the scores to train my reinforcement learning model. Is there any function to get number of passed gates, time and penalties at each step time?
I know I can read them from the log file, but reading a file is time consuming. I need to do it through the API.

madratman, 2019-11-14:
Are you sure it's time consuming? Have you tried reading from the log fie? Reading the file is less overhead than reading via api (over a port and cost of serializing / deserializing). If you need help in how to read the log file, we have sample code in the repo. In the api scenario, you'll essentially call a score api every few ms, which you can do for leading the last line of the log file as well.

kavehkamali, 2019-11-14:
Ok, Thank you. I will try the log file.

surfii3z, 2019-11-15:
@madratman I am also trying to do the same thing. Should I run two separate scripts i.e. baseline_racer.py and log_monitor.py? or Do I make a another thread in the baseline_racer.py for reading the log file?  Thanks

madratman, 2019-11-15:
@surfii3z it's up to you. Yes, you can make a thread in baseline_racer, if you want to have read and use the log in the same script when you're moving the drone.. You can modify the path in the log_monitor script to be AirSimExe/Saved/Logs/RaceLogs - and start the thread after you call simstartrace - so that the log monitor opens the latest file opened