microsoft/AirSim-NeurIPS2019-Drone-Racing #129: Run different instances on different GPUs

yannbouteiller, 2019-11-17:
Hello,

I have just had the bad surprise to find that on a multi-GPU system all my AirSim instances try to run on the same GPU device.

I found this https://github.com/Microsoft/AirSim/issues/1574 , is there any change about this issue or any unsupported way to do this, please?

madratman, 2019-11-17:
Try the method here https://carla.readthedocs.io/en/latest/carla_headless/ or https://github.com/carla-simulator/carla/issues/225
As it's a matter of unreal binary, the instructions therein should help

yannbouteiller, 2019-11-17:
Well, I have tried setting SDL_HINT_CUDA_DEVICE this morning, but this resulted in a segmentation fault when I set it to anything other than 0 even with multiple available GPUs.