microsoft/AirSim-NeurIPS2019-Drone-Racing #132: Recording on multiple cameras at once

collinabidi, 2019-11-20:
I'm having trouble with recording data when running the ```baseline_racer.py``` simulation. I am trying to collect RGB-D data, so I added a second camera to the drone to collect Depth frames that are synchronized to RGB frames.

When I run the basic ```baseline_racer.py``` with modified settings for two cameras, it runs for about 5 seconds and then crashes. I tried to create multiple client and individual threads for polling RGB and Depth separately, but it didn't help.

The issue #7 states that there is a bottleneck issue with Unreal, but I was wondering if there was a way around this that doesn't break the simulation.

madratman, 2019-11-20:
Can you share your setting.json and the simgetimages call snippet so we can repro this? It should not crash with rgb and depth itself.

collinabidi, 2019-11-20:
Before generating settings.json, I added the following code to the ```write_airsim_neurips_baseline_settings_file()``` in ```utils.py```:

```
instance.add_camera(vehicle_name = "drone_1", camera_name = 'depth_cam', relative_pose=Pose(Position(0.25, 0.0, 0.0), Rotation()), image_type = 1, image_width = 320, image_height = 240, fov_horizontal_degrees = 90)
```
Settings.json:
```
{
  "ClockSpeed": 1,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "drone_1": {
      "Cameras": {
        "depth_cam": {
          "CaptureSettings": [
            {
              "FOV_Degrees": 90,
              "Height": 240,
              "ImageType": 1,
              "Width": 320
            }
          ],
          "Pitch": 0.0,
          "Roll": 0.0,
          "X": 0.25,
          "Y": 0.0,
          "Yaw": 0.0,
          "Z": 0.0
        }
      },
      "Pitch": 0.0,
      "Roll": 0.0,
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Yaw": 0.0,
      "Z": 0.0
    },
    "drone_2": {
      "Pitch": 0.0,
      "Roll": 0.0,
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Yaw": 0.0,
      "Z": 0.0
    }
  }
}

```

Here's the callback functions (one for rgb, one for depth) that poll their respective airsim clients.

```
def image_callback(self):
        # get uncompressed fpv cam image
        request = [airsim.ImageRequest('fpv_cam', 0, False, False)]
        response = self.airsim_client_images.simGetImages(request)
        img_1d = np.fromstring(response[0].image_data_uint8, dtype=np.uint8) 
        img_rgb = img_1d.reshape(response[0].height, response[0].width, 3)

        if self.viz_image_cv2:
            cv2.imshow("img_rgb", img_rgb)        
            cv2.waitKey(1)

        # save if specified
        if self.record_cam_data:
            # write image
            timestamp = round(time.time(),6)
            filename = str('color_ims/' + str("{0:.6f}".format(timestamp) + '.png')
            cv2.imwrite(self.output_dir + '/color_ims/' + filename, img_rgb)
            # save timestamp and filename
            self.rgb_data_associations[timestamp] = filename
            
    def depth_image_callback(self):
        # get uncompressed depth cam image
        request = [airsim.ImageRequest('depth_cam', 3, False, False)]
        response = self.airsim_client_depth_images.simGetImages(request)
        img_1d = np.fromstring(response[0].image_data_uint8, dtype=np.uint8) 
        img_depth = img_1d.reshape(response[0].height, response[0].width, 3)

        if self.viz_image_cv2:
            cv2.imshow("img_depth", img_depth)        
            cv2.waitKey(1)

        # save if specified
        if self.record_cam_data:
            timestamp = round(time.time(),3)
            filename = 'depth_ims/' + str("{0:.6f}".format(timestamp) + '.png'
            cv2.imwrite(self.output_dir + '/depth_ims/' + filename, img_depth)
            # save timestamp and filename
            self.depth_data_associations[str(timestamp)] = filename
```

I tried polling a shared client, but that resulted in crashing immediately.

madratman, 2019-11-21:
ok, you need a "capture" not a new "camera"

use this:

```json
{
  "ClockSpeed": 1,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "drone_1": {
      "Cameras": {
        "fpv_cam": {
          "CaptureSettings": [
            {
              "FOV_Degrees": 90,
              "Height": 240,
              "ImageType": 0,
              "Width": 320
            },
            {
              "FOV_Degrees": 90,
              "Height": 240,
              "ImageType": 1,
              "Width": 320
            }
          ],
          "Pitch": 0.0,
          "Roll": 0.0,
          "X": 0.25,
          "Y": 0.0,
          "Yaw": 0.0,
          "Z": 0.0
        }
      },
      "Pitch": 0.0,
      "Roll": 0.0,
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Yaw": 0.0,
      "Z": 0.0
    },
    "drone_2": {
      "Pitch": 0.0,
      "Roll": 0.0,
      "VehicleType": "SimpleFlight",
      "X": 0.0,
      "Y": 0.0,
      "Yaw": 0.0,
      "Z": 0.0
    }
  }
}
```

in general, here's another example https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_tutorial_pkgs/settings/front_stereo_and_center_mono.json

Your python code should be 

```python
responses = self.airsim_client.simGetImages([
     airsim.ImageRequest("fpv_cam", airsim.ImageType.Scene, False, False),
     airsim.ImageRequest("fpv_cam", airsim.ImageType.DepthPlanner, True)
])
if responses[0].height > 0:
    img_rgb_1d = np.fromstring(response[0].image_data_uint8, dtype=np.uint8) 
    img_rgb = img_rgb_1d.reshape(response[0].height, response[0].width, 3)
if responses[1].height > 0:
    depth_img = airsim.list_to_2d_float_array(responses[1].image_data_float, responses[1].width, responses[1].height)
```

`        request = [airsim.ImageRequest('depth_cam', 3, False, False)]` is wrong. you're requesting type 3, but you have type 1 (depthplanar) in json. also it was crashing, as there was no type 0 (scene) image. that settings generator doesn't support captures, but anyway, you can use the above file directly

collinabidi, 2019-11-22:
@madratman ah ok, thank you for the clarification.

I cleaned up my code and used your settings.json file, but it's still crashing and I'm getting a different error post-crash:

```
  File "baseline_racer.py", line 336, in <module>
    main(args)
  File "baseline_racer.py", line 311, in main
    baseline_racer.fly_through_all_gates_at_once_with_moveOnSpline().join()
  File "/home/collin/Projects/AirSim-NeurIPS2019-Drone-Racing/airsimenv/lib/python3.6/site-packages/msgpackrpc/future.py", line 22, in join
    self._loop.start()
  File "/home/collin/Projects/AirSim-NeurIPS2019-Drone-Racing/airsimenv/lib/python3.6/site-packages/msgpackrpc/loop.py", line 22, in start
    self._ioloop.start()
  File "/home/collin/Projects/AirSim-NeurIPS2019-Drone-Racing/airsimenv/lib/python3.6/site-packages/tornado/ioloop.py", line 755, in start
    raise RuntimeError("IOLoop is already running")
RuntimeError: IOLoop is already running
```

madratman, 2019-11-22:
Ensure you are using a single airsim client per thread
see comment in code here https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/blob/master/baselines/baseline_racer.py#L21

collinabidi, 2019-11-23:
I am using that code, so I'm not sure why it would be crashing.