microsoft/AirSim-NeurIPS2019-Drone-Racing #136: Final Windows invalid number of arguments error

Sohojoe, 2019-11-23:
I updated to the final binaries, paks, and code - however, I get the following error when running on windows:

'simGetObjectPose' was called with an invalid number of arguments. Expected: 2, got: 1

```
  File "baseline_racer.py", line 285, in <module>
    main(args)
  File "baseline_racer.py", line 252, in main
    baseline_racer.start_race(args.race_tier)
  File "baseline_racer.py", line 46, in start_race
    self.airsim_client.simStartRace(tier)
  File "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\airsimneurips\client.py", line 155, in simStartRace
    competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneObjects(".*[Gg]ate.*"))]
  File "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\airsimneurips\client.py", line 155, in <listcomp>
    competitor.gate_poses_ground_truth = [self.__internalRandomGoalZone(gate) for gate in sorted(self.simListSceneObjects(".*[Gg]ate.*"))]
  File "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\airsimneurips\client.py", line 177, in __internalRandomGoalZone
    gate_pose = self.__internalGetObjectPose(gate_name)
  File "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\airsimneurips\client.py", line 311, in __internalGetObjectPose
    pose_msgpack = self.client.call('simGetObjectPose', object_name, False)
  File "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'simGetObjectPose' was called with an invalid number of arguments. Expected: 2, got: 1
```

ironclownfish, 2019-11-25:
I have been looking into your issue today, and I'm having trouble reproducing it.
I re-downloaded our binaries and paks from the latest release, and they are functioning correctly for me on Windows.

I notice that the airsim you are using is "C:\Users\ml-2 windows\.conda\envs\airsim\lib\site-packages\airsimneurips\client.py" i.e. it's installed to your anaconda environment. I've had the issue of anaconda defaulting to older clients before -- perhaps that's happening to you? I'd suggest trying a different python3 environment and making absolutely sure you're using the latest client.

In the meantime I'll continue to work on reproducing the bug.

Sohojoe, 2019-11-26:
@ironclownfish - i nuked / rebuilt my conda environment and redownloaded the windows files but having the same problem. It maybe something to do with the conda cache so I will try clearing that

AiltonOliveir, 2020-06-02:
@Sohojoe Do you have success to solve the issue ? We have the same problem in Windows

MajdWardeh, 2020-07-15:
I also have the same problem in Windows.