microsoft/AirSim-NeurIPS2019-Drone-Racing #141: Fresh install and have error

hbzhang, 2020-03-23:
[2020.03.23-15.33.15:497][  0]LogExit: Executing StaticShutdownAfterError
[2020.03.23-15.33.15:575][  0]LogCore: Warning: Unable to statfs('/home/hbzhang/opt/race/AirSim-NeurIPS2019-Drone-Racing-master/AirSim_Final_Round/AirSim/AirSimExe/Saved/Crashes/crashinfo-AirSimExe-pid-17795-004FD20B08D7CF1DFA34D10AF8920BAF/AirSimExe.log'): errno=2 (No such file or directory)
Malloc Size=74119 LargeMemoryPoolOffset=327959 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)

ugurkanates, 2020-03-24:
Would be helpful what files you executed to get this error , also bear in mind this simulation needs to have dedicated VRAM atleast 2GB to work. I worked with NVIDIA GT 840M (2GB Version)

I worked on this simulation for 6 months developing RL framework for it & developing PPO agent along side. So I know how it works mostly would be, can help you if you cant get answer from official developers

madratman, 2020-03-25:
@hbzhang can you share the specs of your machine. As @ugurkanates said, you do need a GPU.

hbzhang, 2020-03-25:
16GB memory. 
[image]

saihv, 2020-03-25:
@hbzhang Please share the complete error output from the terminal. From the log you posted, `StaticShutdownAfterError` typically refers to the part that takes place after the main error.

hbzhang, 2020-03-25:
For somehow, it starts work now for the command. 

./AirSimExe.sh -windowed -opengl4

For the first few times, it does not work. Not sure why....

closed 2020-03-25 by hbzhang

madratman, 2020-03-25:
hmm, weird. you might have had a stray process running in background?  
`ps aux | grep AirSimExe` / `pkill AirSimExe` might be of use. 
I'd try to monitor cpu and gpu usage with `htop` and `nvidia-smi` / [`nvtop`](https://github.com/Syllo/nvtop)

hbzhang, 2020-03-25:
Thanks for the suggestion. I will try.