microsoft/AirSim-NeurIPS2019-Drone-Racing #143: Any way to download/use just the environments?

RitvikN, 2020-03-27:
The environments created for this project are high quality and excellent for training algorithms for drone racing.

However, this competition requires you to use the provided APIs to conduct races and control the drone. I'm interested in trying out some other algorithms, so is it possible to get the environments from the .pak files? Is there any way to load the .pak files into an UE4 project?

yannbouteiller, 2020-03-30:
Actually a colleague in my lab also has an interesting project in which he would need to use the gym environment I made for the 2019 competition, but ideally we would have to slightly modify the .pak files (in order to disable death of the lagging drone in the event of a drone-drone collision, namely). Any hope we can get access to the source code / Unreal project @madratman please?

ugurkanates, 2020-04-02:
I would also love this as well

closed 2020-04-16 by madratman

madratman, 2020-04-29:
https://github.com/microsoft/AirSim/releases/tag/v1.3.1-linux B99, ZJJ, SoccerField are up