microsoft/AirSim-NeurIPS2019-Drone-Racing #147: [Linux] AirSimExe no longer working

AlessandroSaviolo, 2020-05-01:
Hi,

When running ./AirSimExe.sh (either in window mode or without) in Ubuntu 18.04, the simulator crashes without even loading the starting environment.
I have been using Airsim by couple of months and it has always worked without any problem. I've tried to download again this and Airsim repo and to reinstall airsim and airsimneurips packages. But, it keeps crashing without showing anything.
I've also tried to use it with and without rendering but it produced the same error.
The errors that appeared in the terminal are the following:

```
[...]
[2020.05.01-14.24.22:860][  0]r.CustomDepth = "3"
Resizing viewport due to setres change, 1920 x 1080[2020.05.01-14.24.22:860][  0]LogUMG: Display: Widget Class BP_LoadingScreenWidget_C - Loaded Fast Template.
[2020.05.01-14.24.22:861][  0]LogActor: Warning: BP_CameraDirector_C /Game/MainMenu.MainMenu:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
[2020.05.01-14.24.22:863][  0]LogPakFile: New pak file ../../../AirSimExe/Content/Paks/AirSim.pak added to pak precacher.
[2020.05.01-14.24.22:869][  0]LogPakFile: New pak file ../../../AirSimExe/Content/Paks/Soccer_Field.pak added to pak precacher.
ADDDDDING VEHIIIIIIIICCLE drone_1
ADDDDDING VEHIIIIIIIICCLE drone_2
[2020.05.01-14.24.23:205][  0]LogScript: Warning: Script Msg: Attempted to access index 0 from array 'CallFunc_GetAllActorsOfClass_OutActors' of length 0 in '/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C:ExecuteUbergraph_WeatherActor'!
[2020.05.01-14.24.23:207][  0]LogStreaming: Error: Couldn't find file for package /AirSim/HUDAssets/MenuActor requested by async loading code. NameToLoad: /AirSim/HUDAssets/MenuActor
[2020.05.01-14.24.23:207][  0]LogStreaming: Error: Found 0 dependent packages...
[2020.05.01-14.24.23:208][  0]LogTemp: Warning: Warning, WeatherAPI got invalid menu actor class!
[2020.05.01-14.24.23:210][  0]LogTemp: drone_1
drone_1
[2020.05.01-14.24.23:210][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
terminating with uncaught exception of type std::__1::system_error: bind: Address already in use
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=85200 LargeMemoryPoolOffset=216336 
Failed to find symbol file, expected location:
"/home/alessandro/workspace/Binaries/AirSim_Training/AirSimExe/Binaries/Linux/AirSimExe.sym"
[2020.05.01-14.24.23:250][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2020.05.01-14.24.23:250][  0]LogCore: Fatal error!

0x00007f6240079e97 libc.so.6!gsignal(+0xc7)
0x00007f624007b801 libc.so.6!abort(+0x140)
0x0000000006e14dbb AirSimExe!UnknownFunction(0x6c14dba)
0x0000000006e14cd2 AirSimExe!UnknownFunction(0x6c14cd1)
0x0000000006e14af3 AirSimExe!UnknownFunction(0x6c14af2)
0x0000000006e14596 AirSimExe!UnknownFunction(0x6c14595)
0x0000000006e1451f AirSimExe!__cxa_throw(+0x6e)
0x00000000060a295b AirSimExe!_ZN11clmdep_asio6detail15throw_exceptionINSt3__112system_errorEEEvRKT_(+0x4a)
0x00000000060a28dc AirSimExe!_ZN11clmdep_asio6detail14do_throw_errorERKNSt3__110error_codeEPKc(+0x3b)
0x00000000060a2892 AirSimExe!_ZN11clmdep_asio6detail11throw_errorERKNSt3__110error_codeEPKc(+0x31)
0x00000000060a2249 AirSimExe!_ZN11clmdep_asio21basic_socket_acceptorINS_2ip3tcpENS_23socket_acceptor_serviceIS2_EEEC2ERNS_10io_serviceERKNS1_14basic_endpointIS2_EEb(+0x258)
0x00000000060a0312 AirSimExe!_ZN3rpc6server4implC2EPS0_t(+0xb1)
0x000000000609d4f0 AirSimExe!_ZN3rpc6serverC1Et(+0x7f)
0x00000000025c021c AirSimExe!UnknownFunction(0x23c021b)
0x000000000265d4a9 AirSimExe!UnknownFunction(0x245d4a8)
0x000000000269d7a3 AirSimExe!UnknownFunction(0x249d7a2)
0x00000000026868df AirSimExe!UnknownFunction(0x24868de)
0x000000000268542a AirSimExe!UnknownFunction(0x2485429)
0x0000000004a3ebb2 AirSimExe!UnknownFunction(0x483ebb1)
0x00000000057a722d AirSimExe!UnknownFunction(0x55a722c)
0x0000000004ed882c AirSimExe!UnknownFunction(0x4cd882b)
0x0000000005780f79 AirSimExe!UnknownFunction(0x5580f78)
0x0000000005686226 AirSimExe!UnknownFunction(0x5486225)
0x00000000056824c0 AirSimExe!UnknownFunction(0x54824bf)
0x0000000004ea9ddc AirSimExe!UnknownFunction(0x4ca9ddb)
0x0000000004e9526e AirSimExe!UnknownFunction(0x4c9526d)
0x000000000257f86c AirSimExe!UnknownFunction(0x237f86b)
0x0000000002589e67 AirSimExe!UnknownFunction(0x2389e66)
0x0000000005da982f AirSimExe!UnknownFunction(0x5ba982e)
0x00007f624005cb97 libc.so.6!__libc_start_main(+0xe6)
0x0000000002579029 AirSimExe!UnknownFunction(0x2379028)

[2020.05.01-14.24.23:400][  0]LogExit: Executing StaticShutdownAfterError
[2020.05.01-14.24.23:434][  0]LogCore: Warning: Unable to statfs('/home/alessandro/workspace/Binaries/AirSim_Training/AirSimExe/Saved/Crashes/crashinfo-AirSimExe-pid-13033-006ADEAD08D7EDEC1BFF8FEA3A21BB67/AirSimExe.log'): errno=2 (No such file or directory)
Malloc Size=41717 LargeMemoryPoolOffset=258069 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

Thanks in advance for any help!

AlessandroSaviolo, 2020-05-02:
The problem was Ubuntu not killing the process correctly. Had to use system monitor to find it and kill it. Closing the issue.

closed 2020-05-02 by AlessandroSaviolo

skamano, 2020-06-09:
May I ask what the process name was? I'm running into this exact same issue but I haven't been able to resolve it.

AlessandroSaviolo, 2020-06-09:
Use the command `pkill AirSimExe`. Hope it helps!