microsoft/AirSim-NeurIPS2019-Drone-Racing #149: Race finished after reaching first gate

AlessandroSaviolo, 2020-05-04:
Hello,

I'm running multiple instances of Airsim in sequence. Each time I run an instance, I reload the level (to avoid a problem in the stability of the drone by just resetting the race without reloading the level). But, I'm having problems with the simulator interrupting the race after I reach the first gate, outputting that the drone has finished the race. This behaviour does happen every even iteration excluded 0 (2, 4, 6, 8, ...).

Since I monitor the log file for the timings, this is a big problem because it cuts the log file:

```
[...]
drone_1 time 1758 gates_passed 1
drone_1 time 1758 gates_missed 11
drone_1 time 1758 finished 1
```
I'm using Soccer Field Easy and Tier 1 for my simulations.

Can you point me the part of code where the simulator detects the last gate and interrupts the race?

Thanks in advance

surfii3z, 2020-05-07:
Hi, not sure though, I think I use to have similar problems.

Maybe this could help

[https://github.com/microsoft/AirSim-NeurIPS2019-Drone-Racing/issues/104#issuecomment-546087098](url)

I change to reset the race using this function and it was working for sometimes (like 0.5 hrs) before the simulation crashed

surfii3z, 2020-05-07:
By the way, during my project, I noticed that if drone_2 passed the gate faster than drone_1 (which you control, by default). The log won't report the event of drone_1 passing the gate

AlessandroSaviolo, 2020-05-07:
Hi, thanks for the reply!

I'm trying the same workaround with simResetRace. Before I had a very unstable behavior on the podium after restarting the race multiple times (already from the second iteration I could see the drone shaking a bit), but I've probably fixed it by resetting also the airsim client.

And yes, I'm experiencing the same problems with the simulator freezing. I'll give a look to that issue, thanks again :)

AlessandroSaviolo, 2020-05-22:
Update: I've successfully avoided the problem of this post by just using simResetRace and resetting the client.

> By the way, during my project, I noticed that if drone_2 passed the gate faster than drone_1 (which you control, by default). The log won't report the event of drone_1 passing the gate

I'll open a new issue about this, since it's creating a lot of problems for my tests.